SonixQMK/qmk_firmware #15: [Aukey KM-G12] Add support

maxill1, 2021-03-29:
## Feature Request Type

- [ x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

VID=0C45 and PID=8006, so it should be a sonix chip.

Sadly aukey doesn't provide much info on this keyboard but really looks like a clone of Redragon K556 (full layout 104/105 keys).

I'm not sure how to enter in bootloader mode without full disassembly the keyboard so I can't test right now the K556 firmware. Is there any test I could do to extract useful infos?

Krutonium, 2021-04-04:
Your best bet is to tear the keyboard down and take a picture of the chip that's running the show, that will tell us a lot.

maxill1, 2022-01-05:
I finally found time to tear down the keyboard.

lsusb result:
`ID 0c45:8006 Microdia Dual Mode Camera (8006 VGA)`

Chip: **HFD2201KBA**
<details>

[image]

</details>

Board: **GK4105M-RGB-HFD** (switches are soldered directly on board so i can't take pictures of the front board right now)
<details>

[image]

[image]

</details>

Brand: **[Aukey KM-G12](https://it.aukey.com/products/km-g12-mechanical-keyboard-blue-switches)** (according to [this OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/1992) issue it may to be an OEM clone of "Sandberg Firestorm Mech Keyboard")
Layout: 100% 
<details>

[image]
[image]

</details>

closed 2022-01-22 by dexter93