SonixQMK/qmk_firmware #70: [Bug] ISO K552 bugs found

daniarla, 2021-08-02:
## Describe the Bug

I flashed the latest firmware available and I found various bugs. This is a documentations of them.

1. The NKRO no longer works and you only get 5-kro.
2. The right alt does not work.
3. It does not work with via (But i suppose that that is deactivated on purpose)

## System Information

 - Keyboard:  ISO K552
   - Revision (if applicable):   rev 2
 - Operating system: Arch Linux /  Windows 
 - QMK Firmware version: Last compiled firmware from github actions 
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

My model is the ISO version and it could cause key-map related bugs.

CalcProgrammer1, 2021-08-04:
Indicator LEDs are set by the OS, they are not automatically toggled on the keyboard side.  If it works in GRUB but not in Linux, the firmware is fine.   Something in Linux is preventing the indicator from toggling on.

daniarla, 2021-08-05:
I suspected that but I wanted to get a 2nd opinion on it just to be sure!

I edited my previous comments reflect only the 3 first problems!

daniarla, 2021-08-09:
I corrected my first message because the key not working was the alt-gr, not the alt as my keycap set has no alt-gr keycap and I used and alt.
This is a ISO only key.

manuhank, 2022-05-08:
@daniarla alt_gr its working fine for me. would you test using my branch on this PR?https://github.com/SonixQMK/qmk_firmware/pull/264
If you don't  know how to compile, pinkg me and I can send you the bin.

daniarla, 2022-05-09:
@manuhank alt_gr was also working fine for me, it was a problem with the software that I used for testing.
Btw, do you have N-key rollover working? I remember  that being an issue with qmk.

manuhank, 2022-05-09:
@daniarla using this web https://www.mechanical-keyboard.org/, I can press up to 6 letters/numbers. It also adds up the modifiers key.
Not sure if that NKRO, 6KRO, or what.

daniarla, 2022-05-09:
@manuhank Yeah with qmk you get then 6kro, but with the original firmware you get NKey. I'm not sure why.
In that web I managed to get like 28 different keys detected.

manuhank, 2022-05-09:
HAve you tried adding:

/* Enable NKRO - Up to 248 keys at the same time. */
#define FORCE_NKRO

to the config?

manuhank, 2022-05-09:
Just tested and seems working fine. here you have a default build 
 with NKRO enabled.https://1drv.ms/u/s!AvQM1RvGBOHEicNE50z6i7gYD9CDQA?e=iRveNg
 Enjoy

manuhank, 2022-05-09:
If you too confirm this works correctly, I would ask @CalcProgrammer1  what's the policy over this. should I rise a PR for this to be the default behavior, or it would be considered a personal config, and thus, not of interest for the repo. In that last case, this issue should be closed.

daniarla, 2022-05-31:
@manuhank Im sorry for the delayed response but I had exams and a lot of things to work on.
I just tried your firmware and I can confirm that NKRO works perfectly!

closed 2023-10-21 by dexter93