SonixQMK/qmk_firmware #71: [Bug] RGB on unified key matrix will not work for PWM B

dexter93, 2021-08-03:
## Describe the Bug

PFPA is not declared for boards requiring PWM B side control.
That results in affected columns missing rgb control and staying off
Models affected: keychron k4,c1 and possibly k8

`    // PWM0-2 is B for the k4
    SN_PFPA->CT16B1 = 0x00000007;
`
this is the old way of tackling this. It might be worth seeing if we can handle that in chibios level instead of setting the registers directly in matrix code.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

dexter93, 2021-08-04:
> Can we reasonably assume that one PWM output is only going to be used once on either A or B? It wouldn't be too difficult to check for the PWM B pins in the pin list and then remapping the port during setup.

We haven't found any boards so far that use both at the same time, so it would probably be the way to go for a clean approach

CalcProgrammer1, 2021-08-07:
Pushed a change that adds the PWM B pins to the matrix initialization.  I think this will resolve this issue.  Please test.

CalcProgrammer1, 2021-08-07:
Fix confirmed working at least on C1 on Discord.  Closing this issue.

closed 2021-08-07 by CalcProgrammer1