SonixQMK/qmk_firmware #78: [Bug] orgb + shared ep + force nkro + media keys = kb unresponsive after restarts

stdvar, 2021-08-07:


stdvar, 2021-08-07:
Check USB lld driver + shared ep code to identify potential bug

vjdato21, 2021-08-08:
### Initial findings

**Shared EP + NKRO + FORCED NKRO** = Responsive keyboard even with RGB programs hooked via direct mode. Keyboard though doesn't register keystrokes when PC reboot is done. 

**NKRO + FORCED NKRO** = Recognizes more than 6 keystrokes but, keystrokes do not register properly when used with RGB programs hooked via direct mode. 

**NKRO only** = only recognizes 6 keystrokes max, it can also have performance problems depending on your RGB settings/program.

vjdato21, 2021-08-10:
### Possible cause identified.
Keyboard may not be detected if the endpoints are shared in some BIOSes.

vjdato21, 2021-08-12:
https://beta.docs.qmk.fm/developing-qmk/qmk-reference/config_options

> ### Possible cause identified.
> Keyboard may not be detected if the endpoints are shared in some BIOSes.

closed 2023-10-21 by dexter93