SonixQMK/qmk_firmware #80: [Bug] Suspend works but resume doesn't.

l33tlinuxh4x0r, 2021-08-10:
Trying to suspend the RGB on PC sleep/shutdown and resume on wake/boot.

This part works. It turns off the RGB
```
void suspend_power_down_user(void) {
    rgb_matrix_set_suspend_state(true);
}

```

This part does not work
```
void suspend_wakeup_init_user(void) {
    rgb_matrix_set_suspend_state(false);
}
```

`#define RGB_DISABLE_WHEN_USB_SUSPENDED true`
is set in config.h

## System Information

 - Keyboard:  Redragon K556
   - Revision (if applicable):  
 - Operating system: Arch Linux
 - AVR GCC version: avr-gcc (GCC) 11.2.0

 - ARM GCC version: arm-none-eabi-gcc (Arch Repository) 11.2.0

 - QMK Firmware version: 0.7.101

 - Any keyboard related software installed? 
   - [ ] AutoHotKey No
   - [ ] Karabiner No
   - [ ] Other: OpenRGB, VIA

## Additional Context

IsaacDynamo, 2021-09-07:
I think I have the same issue on my Keychron C1 White. suspend_power_down_user() gets called but suspend_wakeup_init_user() not always.

I have done the following test with Window 10, Start -> Power -> Sleep.
1. Wakeup via other keyboard, suspend_wakeup_init_user() is never called. backlight stays off.
2. Wakeup via C1, suspend_wakeup_init_user() get called but when the key that is used to wake up is released, suspend_power_down_user() gets called, and backlight turns off.
3 Wakeup via C1 by spamming a key. This calls suspend_wakeup_init_user() multiple times, backlight turns on and off, but in the end, when the PC is woken-up, the backlight is sometimes on.
In all the three cases the keyboard functions after the wakeup it is just the backlight that sometimes doesn't turn on.

l33tlinuxh4x0r, 2021-10-18:
hmm It looks like this works for one or 2 reboots then fails... Looking into it further.

l33tlinuxh4x0r, 2021-10-18:
Ok I successfully tested it 3-4 times in a row... Seems that one tiny change fixes it. Just like above except that keymap should have this added instead.

```
void suspend_power_down_kb(void) {
   rgb_matrix_set_suspend_state(true);
}

void suspend_wakeup_init_user(void) {
   rgb_matrix_set_suspend_state(false);
}
```

config.h
```
#define RGB_DISABLE_WHEN_USB_SUSPENDED true
```

l33tlinuxh4x0r, 2021-10-20:
I have narrowed down the issue a bit... The sleep always works it is the resume that is broken... 
```
void suspend_wakeup_init_user(void) {
```
I have also tried 
```
void suspend_power_down_user(void) {
    rgb_matrix_disable();
}

void suspend_wakeup_init_user(void) {
    rgb_matrix_enable();
}
```
which has the exact same issue, however it avoids needing to define RGB_DISABLE_WHEN_USB_SUSPENDED

l33tlinuxh4x0r, 2021-10-20:
I finally got this fixed/working. I tested with several sleep/reboot cycles. All that needs to be done is to add the following to keymap.c

```
void suspend_power_down_user(void) {
    rgb_matrix_disable();
}

void housekeeping_task_user(void) {
    rgb_matrix_enable();
}
```

This would probably work using the RGB_DISABLE_WHEN_USB_SUSPENDED method but I haven't tested that yet... The important part is houskeeping_task_user(void)

Can someone please test to confirm?

vjdato21, 2022-01-02:
> As per discord... The above breaks the RGB off keybind. The following allows the RGB off keybind to work and also the RGB to suspend and resume correctly.
> 
> Add this to keymap.c
> 
> ```
> void suspend_power_down_kb(void) {
>     rgb_matrix_set_suspend_state(true);
> }
> 
> void housekeeping_task_user(void) {
>     rgb_matrix_set_suspend_state(false);
> }
> ```
> 
> Add this to config.h
> 
> ```
> /* Disable RGB while USB is sleeping */
> #define RGB_DISABLE_WHEN_USB_SUSPENDED true
> ```

Hmm... For some reason, this kinda introduces a new issue where the keyboard my self-brick itself. I suppose it's writing on the EEPROM when disabling or enabling RGB matrix during USB suspend.

l33tlinuxh4x0r, 2022-01-27:
This seems to be fixed in the sn32_master branch by adding only the following line to the config.h file

```
#define RGB_DISABLE_WHEN_USB_SUSPENDED true
```

l33tlinuxh4x0r, 2022-02-10:
This issue is fixed for me on latest branch with the above comment. I'm closing this bug report.

closed 2022-02-10 by l33tlinuxh4x0r