SonixQMK/qmk_firmware #81: [Bug] Default rgb animation

dexter93, 2021-08-12:
Default RGB animation is rainbow

However, when the usb driver kicks in, it switches to red breathing. You can observe this behavior by unplugging the keyboard and switching to battery ( bluetooth) mode.

Need to figure out this inconsistency and see if it is our code, or QMK upstream behavior

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

dexter93, 2021-11-17:
Bug fixed with https://github.com/SonixQMK/qmk_firmware/pull/150

Coding-Kiwi, 2021-12-26:
I'm at the latest version with the changes of #150 merged in, I can still reproduce the issue on Win10 with a keychron k4v2 iso rgb

process:

1) switch the animation to rgb spiral mode
2) power down pc -> rgb is off
3) power on pc -> rgb is in the saved mode (rgb spiral), but then switches to breathing mode by itself

I also tried the changes suggested [here](https://github.com/SonixQMK/qmk_firmware/issues/80#issuecomment-948231491)

closed 2023-10-21 by dexter93