SonixQMK/qmk_firmware #86: [Bug] LED Status on C1 Hotswap RGB

vjdato21, 2021-08-17:
## Describe the Bug

- Toggling num lock affects the **Windows Mode** status LED.
- If the keyboard was set to **Mac Mode** and the num lock was toggled, the **Mac Mode** status LED turns off **Windows Mode** turns on instead.
- **Mac Mode** is now set as scroll lock status LED. (Tested it by pressing scroll lock, this issue wasn't present before the unification of matrix.)

## System Information

 - Keyboard:  Keychron C1 Hotswap RGB
   - Revision (if applicable):  
 - Operating system: Windows 10 21H1
 - AVR GCC version: 8.4.0

 - ARM GCC version: 10.1.0

 - QMK Firmware version: 0.7.101

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
[https://streamable.com/na9g4b](url)

IsaacDynamo, 2021-09-07:
Had similar issues with my C1 White and fixed it. Maybe you can do something similar.
https://github.com/IsaacDynamo/qmk_firmware/blob/keychron_c1_white/keyboards/keychron/c1_white/keymaps/default/keymap.c

vjdato21, 2021-09-16:
A fix is coming...

closed 2021-09-16 by vjdato21