SonixQMK/qmk_firmware #89: [Feature Request] Support for the Redragon K551 Mitra RGB Keyboard

carbonatek, 2021-08-19:
This keyboard is very similar to the K552, which is very likely just a tenkeyless version of the K551 under the hood (unless not). Should be easy to implement. Also, I have it, and it isn't supported, so that's the main reason I'm pitching this suggestion here. :-)
Not sure if this might be of any use, but the vendor ID is 0C45 and the device ID is 5004.

KinteLiX, 2021-09-04:
Hi, the k556 firmware works flawlessly on my k551 (I think they are practically the same keyboard if we are talking about the RGB version of the k551).

KinteLiX, 2021-10-17:
> Hi, the k556 firmware works flawlessly on my k551 (I think they are practically the same keyboard if we are talking about the RGB version of the k551).

Not exactly flawless, the three lights indicating caps lock, num lock, and scroll lock state are kinda bugged. Sometimes pressing caps lock will toggle scroll lock LED, most of the time the LED isn't in sync with the key, pressing those keys won't always toggle the LED (aka caps lock could be on but the led would be off). I'm not talking about the LEDs on the switches themselves, only about the three red lights on the top right. Apart from that everything works.

zyw80, 2022-12-23:
I'm also looking for the K551 port

closed 2023-07-22 by github-actions[bot]