SonixQMK/qmk_firmware #152: [Bug] Keychron faking Apple Keyboard

dexter93, 2021-11-18:
This is causing a mess and potential legal issues.

TODO:  fix the ID's by Model/Layout following this convention
|   VID  	|   PID  	| Model 	| Layout 	|   PID  	| Model 	| Layout 	|
|:------:	|:------:	|:-----:	|:------:	|:------:	|:-----:	|:------:	|
| 0x3434 	| 0xFE00 	|   K1  	|  ANSI  	| 0xFE20 	|   C1  	|  ANSI  	|
| 0x3434 	| 0xFE01 	|       	|   ISO  	| 0xFE21 	|       	|   ISO  	|
| 0x3434 	| 0xFE02 	|   K2  	|  ANSI  	| 0xFE22 	|   C2  	|  ANSI  	|
| 0x3434 	| 0xFE03 	|       	|   ISO  	| 0xFE23 	|       	|   ISO  	|
| 0x3434 	| 0xFE04 	|   K3  	|  ANSI  	| 0xFE24 	|   C3  	|  ANSI  	|
| 0x3434 	| 0xFE05 	|       	|   ISO  	| 0xFE25 	|       	|   ISO  	|
| 0x3434 	| 0xFE06 	|   K4  	|  ANSI  	| 0xFE26 	|   C4  	|  ANSI  	|
| 0x3434 	| 0xFE07 	|       	|   ISO  	| 0xFE27 	|       	|   ISO  	|
| 0x3434 	| 0xFE08 	|   K5  	|  ANSI  	| 0xFE28 	|   C5  	|  ANSI  	|
| 0x3434 	| 0xFE09 	|       	|   ISO  	| 0xFE29 	|       	|   ISO  	|
| 0x3434 	| 0xFE0A 	|   K6  	|  ANSI  	| 0xFE2A 	|   C6  	|  ANSI  	|
| 0x3434 	| 0xFE0B 	|       	|   ISO  	| 0xFE2B 	|       	|   ISO  	|
| 0x3434 	| 0xFE0C 	|   K7  	|  ANSI  	| 0xFE2C 	|   C7  	|  ANSI  	|
| 0x3434 	| 0xFE0D 	|       	|   ISO  	| 0xFE2D 	|       	|   ISO  	|
| 0x3434 	| 0xFE0E 	|   K8  	|  ANSI  	| 0xFE2E 	|   C8  	|  ANSI  	|
| 0x3434 	| 0xFE0F 	|       	|   ISO  	| 0xFE2F 	|       	|   ISO  	|
| 0x3434 	| 0xFE10 	|   K9  	|  ANSI  	| 0xFE30 	|   C9  	|  ANSI  	|
| 0x3434 	| 0xFE11 	|       	|   ISO  	| 0xFE31 	|       	|   ISO  	|
| 0x3434 	| 0xFE12 	|  K10  	|  ANSI  	| 0xFE32 	|  C10  	|  ANSI  	|
| 0x3434 	| 0xFE13 	|       	|   ISO  	| 0xFE33 	|       	|   ISO  	|
| 0x3434 	| 0xFE14 	|  K11  	|  ANSI  	| 0xFE34 	|  C11  	|  ANSI  	|
| 0x3434 	| 0xFE15 	|       	|   ISO  	| 0xFE35 	|       	|   ISO  	|
| 0x3434 	| 0xFE16 	|  K12  	|  ANSI  	| 0xFE36 	|  C12  	|  ANSI  	|
| 0x3434 	| 0xFE17 	|       	|   ISO  	| 0xFE37 	|       	|   ISO  	|
| 0x3434 	| 0xFE18 	|  K13  	|  ANSI  	| 0xFE38 	|  C13  	|  ANSI  	|
| 0x3434 	| 0xFE19 	|       	|   ISO  	| 0xFE39 	|       	|   ISO  	|
| 0x3434 	| 0xFE1A 	|  K14  	|  ANSI  	| 0xFE3A 	|  C14  	|  ANSI  	|
| 0x3434 	| 0xFE1B 	|       	|   ISO  	| 0xFE3B 	|       	|   ISO  	|
| 0x3434 	| 0xFE1C 	|  K15  	|  ANSI  	| 0xFE3C 	|  C15  	|  ANSI  	|
| 0x3434 	| 0xFE1D 	|       	|   ISO  	| 0xFE3D 	|       	|   ISO  	|
| 0x3434 	| 0xFE1E 	|  K16  	|  ANSI  	| 0xFE3E 	|  C16  	|  ANSI  	|
| 0x3434 	| 0xFE1F 	|       	|   ISO  	| 0xFE3F 	|       	|   ISO  	|

IsaacDynamo, 2021-11-18:
What is the issue with using same VID and PID as Keychron stock firmware?

dexter93, 2021-12-26:
> What is the issue with using same VID and PID as Keychron stock firmware?

See https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4 . Keychron is moving away from it for new boards too. `0x3434` is the registered Keychron VID, and those listed are the PID's allocated for their Sonix lineup