SonixQMK/qmk_firmware #232: K8 ISO

mtandrup, 2022-02-08:
I'm trying to make an ISO firmware for my Keychron K8.

This is all new to me, but is the K8 ISO folder not missing an info.json file?

Kind regards
Michael Tandrup
Denmark

tbowmo, 2022-02-09:
No, info.json is only used when using qmk configurator, which we are not part of yet. You have to build it yourself.

Normally you should use the branch sn32_master (it's the newest and shiniest). But if I remember right, there is a bug for the iso rgb, that is fixed in #205. That pr also brings support for mac / win layers, so you could check that out. (or wait a bit until it's merged)

To build your own, you should use something like:
`qmk compile -kb keychron/k8/rgb/iso -km iso`

mtandrup, 2022-02-09:
Thank you.

I was having trouble compiling the ISO layout, not the ANSI layout, which is why I thought it had something to do with the info.json, as it was the only difference I could find between the ANSI and ISO layout.

I thought I tried every possible command besides the one you just suggested.
I think it was the /iso at the end of the path that did the trick.

It's working without issues now.

Thanks again :-)

closed 2022-02-09 by dexter93

davecarrijo, 2022-06-15:
> Thank you.
> 
> I was having trouble compiling the ISO layout, not the ANSI layout, which is why I thought it had something to do with the info.json, as it was the only difference I could find between the ANSI and ISO layout.
> 
> I thought I tried every possible command, but apparently not the one you just suggested. I think it was the /iso at the end of the path that did the trick.
> 
> It's working without issues now.
> 
> Thanks again :-)

Via is working?

doodhout, 2022-09-22:
As far as I understood: original keychron K series keyboards (excluding the newer K8 Pro) do not have onboard memory for changing the keymappings on-the-go, so **no** VIA will not work.

IIRC the only way to remap the keys on original Keychron K-series keyboards is by tripping the bootloader and flashing a precompiled firmware onto it where the keys are mapped to your liking. However to make any changes to this mapping, you will need to edit the pre-compile mapping file, compile it again and flash this new resulting firmware file again before you can use the changed key mapping.

TL;DR: most keychron K-series keyboards have no onboard memory, so VIA will not work, however flashing compiled QMK firmware with custom mapping _does_ work. Be aware that you might/will lose bluetooth functionality when using the QMK firmware.

tbowmo, 2022-09-22:
@doodhout Don't tell my k8 iso (non-pro) that it can't do VIA.. It works like a charm. It's correct that there is no eeprom for storing via settings in the keyboard. But we have an eeprom emulation in flash, so we can still store keyboard / via settings there.

The RGB version has a larger chip (flash memory), than the non-rgb versions, so that is the preferred variant for hacking.

davecarrijo, 2022-09-25:
> @doodhout Don't tell my k8 iso (non-pro) that it can't do VIA.. It works like a charm. It's correct that there is no eeprom for storing via settings in the keyboard. But we have an eeprom emulation [...]

crying in the ANSI variant, I just can't. Tried so many times and ways. Via refuses to work with the ANSI one (RGB).

doodhout, 2022-10-04:
@tbowmo I'm sorry, I was not aware, I figured VIA is only used for making on-the-fly changes.

What does it mean exactly to have eeprom emulation in flash? That you can edit your layout but have to do some trickery to "store" it on the keyboard's memory?

As a sidequest: so you're running QMK on your K8 ISO. Do you have bluetooth working? I have that exact same keyboard and I use bluetooth daily, but would very much like to also have the flexibility of QMK.

doodhout, 2022-10-26:
Since the K8 Pro (with proper QMK support _and_ bluetooth support) is available, it's almost moot to try and properly add QMK to the K8 without losing bluetooth.

tbowmo, 2022-10-26:
Bluetooth is coming very soon (tm) 
I've had it working a while ago, with bleeding edge branches.. People have started working on it again, and are preparing PRs to bring it to the latest develop branches.

doodhout, 2022-10-26:
That would genuinely chuff me to bits, excellent news! Thanks for the update.

Drowze, 2024-06-12:
Hey @tbowmo, did you get any more progress on getting bluetooth to work? 😃

tbowmo, 2024-06-12:
Not at the moment, I've switched to a q11 as main kb in the meantime (no wireless), I still have my k8 somewhere, but haven't used it for half a year or so.