SonixQMK/qmk_firmware #261: State of SonixQMK? Support needed?

cmorty, 2022-04-25:
Hey, what is the state of this Project? The last activity is 1,5 months ago and there are 37 MRs piling up. I know this is solely a fun project, but is there any way it can be supported, to free resources or improve motivation? I can offer review and some development resources, but I need something, too (too many projects that got abandoned despite people putting efforts into it). ;)  Maybe a roadmap?

dexter93, 2022-05-01:
Too few devs and limited time. Cleanup also needed in places since we're trying to get things merged to upstream repos, so the trees are a mess. Timeline is a bit vague now, but there's an order of things to be done. Join up on discord to discuss it further if you want to help :)

crazyquark, 2022-09-04:
how can I help? I have a GMMK TKL rev2 for which I compiled a firmware from this repo.

closed 2023-07-22 by github-actions[bot]