SonixQMK/qmk_firmware #267: Keychron K7 Gateron RGB

daltona, 2022-06-12:
Current sn32_master, despite of supporting the keychron K7, is not working on my brand new keyboard. 

## System Information

 - Keyboard:  
   Keychron K7 Gateron mechanical (not optical) (K004-3-RGB 48RGB V1.1 20211125)

## Additional Context

After probin the key matrix and the led matrix pins to microcontroller, I managed to start using it as a keyboard.

What should I do to properly add this code to the repository ( my understanding i that current changes should only apply to the non-optical version of the K7 keyboard)

Here is the changes that make my device work as a keyboard:

`
diff --git a/keyboards/keychron/k7/keymaps/ansi/keymap.c b/keyboards/keychron/k7/keymaps/ansi/keymap.c
index 6f9dddab48..7d368d18e1 100644
--- a/keyboards/keychron/k7/keymaps/ansi/keymap.c
+++ b/keyboards/keychron/k7/keymaps/ansi/keymap.c
@@ -28,24 +28,24 @@ enum layer_names {
 
 const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
 
- /*      Row:         0          1          2          3        4         5        6         7        8        9          10         11         12         13      */
-      [_BASE] = { {   KC_GESC,   KC_1,      KC_2,      KC_3,    KC_4,     KC_5,     KC_6,     KC_7,    KC_8,    KC_9,      KC_0,      KC_MINS,   KC_EQL,    KC_BSPC },
-                  {   KC_TAB,    KC_Q,      KC_W,      KC_E,    KC_R,     KC_T,     KC_Y,     KC_U,    KC_I,    KC_O,      KC_P,      KC_LBRC,   KC_RBRC,   KC_BSLASH },
-                  {   KC_CAPS,   KC_A,      KC_S,      KC_D,    KC_F,     KC_G,     KC_H,     KC_J,    KC_K,    KC_L,      KC_SCLN,   KC_QUOT,   KC_NO,     KC_ENT },
-                  {   KC_LSFT,   KC_NO,     KC_Z,      KC_X,    KC_C,     KC_V,     KC_B,     KC_N,    KC_M,    KC_COMM,   KC_DOT,    KC_SLSH,   KC_RSFT,   KC_UP },
-                  {   KC_LCTL,   KC_LGUI,   KC_LALT,   KC_SPC,  KC_RCTRL, MO(_FN1), MO(_FN2), KC_LEFT, KC_DOWN, KC_RIGHT,  KC_PGDN,   KC_PGUP,   KC_HOME,   KC_DEL }
+ /*      Row:         0          1          2          3        4         5        6         7        8        9          10         11         12         13         14        */
+      [_BASE] = { {   KC_GESC,   KC_1,      KC_2,      KC_3,    KC_4,     KC_5,    KC_6,     KC_7,    KC_8,    KC_9,      KC_0,      KC_MINS,   KC_EQL,    KC_BSPC,   RGB_MOD   },
+                  {   KC_TAB,    KC_Q,      KC_W,      KC_E,    KC_R,     KC_T,    KC_Y,     KC_U,    KC_I,    KC_O,      KC_P,      KC_LBRC,   KC_RBRC,   KC_BSLASH, KC_HOME   },
+                  {   KC_CAPS,   KC_A,      KC_S,      KC_D,    KC_F,     KC_G,    KC_H,     KC_J,    KC_K,    KC_L,      KC_SCLN,   KC_QUOT,   KC_NO,     KC_ENT,    KC_PGUP   },
+                  {   KC_LSFT,   KC_NO,     KC_Z,      KC_X,    KC_C,     KC_V,    KC_B,     KC_N,    KC_M,    KC_COMM,   KC_DOT,    KC_SLSH,   KC_RSFT,   KC_UP,     KC_PGDOWN },
+                  {   KC_LCTL,   KC_LGUI,   KC_LALT,   KC_NO,   KC_NO,    KC_NO,   KC_SPC,   KC_NO,   KC_NO,   KC_RCTRL,  MO(_FN1),  MO(_FN2),  KC_LEFT,   KC_DOWN,   KC_RGHT   }
                 },
-      [_FN1] = {  {   KC_GESC,   KC_SLCK,   KC_PAUS,   KC_APP,  KC_NO,    RGB_VAD, RGB_VAI,  KC_MPRV, KC_MPLY, KC_MNXT,   KC_MUTE,   KC_VOLD,   KC_VOLU,   KC_NO    },
-                  {   KC_NO,     RGB_M_P,   RGB_M_B,   RGB_M_R, RGB_M_SW, RGB_M_T, RGB_SPD,  RGB_SPI, KC_NO,   KC_NO,     KC_INS,    KC_DEL,    KC_END,    KC_NO  },
-                  {   KC_NO,     KC_VOLD,   KC_VOLU,   KC_NO,   KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   KC_NO,     RGB_SAD,   RGB_SAI,   KC_NO,     KC_NO    },
-                  {   KC_NO,     KC_NO,     KC_NO,     RGB_TOG, KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   RGB_RMOD,  RGB_MOD,   KC_NO,     KC_NO,     RGB_VAI  },
-                  {   KC_GRV,    KC_NO,     KC_NO,     KC_NO,   KC_NO,    KC_NO,   RESET,    KC_NO,   KC_NO,   KC_NO,     KC_NO,     KC_NO,     RGB_HUD,   RGB_VAD    }
+      [_FN1] = {  {   KC_GESC,   KC_SLCK,   KC_PAUS,   KC_APP,  KC_NO,    RGB_VAD, RGB_VAI,  KC_MPRV, KC_MPLY, KC_MNXT,   KC_MUTE,   KC_VOLD,   KC_VOLU,   KC_NO,     KC_PGUP    },
+                  {   KC_NO,     RGB_M_P,   RGB_M_B,   RGB_M_R, RGB_M_SW, RGB_M_T, RGB_SPD,  RGB_SPI, KC_NO,   KC_NO,     KC_INS,    KC_DEL,    KC_END,    KC_NO,     KC_PGDOWN  },
+                  {   KC_NO,     KC_VOLD,   KC_VOLU,   KC_NO,   KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   KC_NO,     RGB_SAD,   RGB_SAI,   KC_NO,     KC_NO,     KC_PGUP    },
+                  {   KC_NO,     KC_NO,     KC_NO,     RGB_TOG, KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   RGB_RMOD,  RGB_MOD,   KC_NO,     KC_NO,     RGB_VAI,   KC_PGDOWN  },
+                  {   KC_GRV,    KC_NO,     KC_NO,     KC_NO,   KC_NO,    KC_NO,   RESET,    KC_NO,   KC_NO,   KC_NO,     KC_NO,     KC_NO,     RGB_HUD,   RGB_VAD,   RGB_HUI    }
                },
-      [_FN2] = {  {   KC_GESC,   KC_F1,     KC_F2,     KC_F3,   KC_F4,    KC_F5,   KC_F6,    KC_F7,   KC_F8,   KC_F9,     KC_F10,    KC_F11,    KC_F12,    KC_NO    },
-                  {   KC_NO,     RGB_M_P,   RGB_M_B,   RGB_M_R, RGB_M_SW, RGB_M_T, RGB_SPD,  RGB_SPI, KC_NO,   KC_NO,     KC_NO,     KC_NO,     KC_NO,     KC_NO  },
-                  {   KC_NO,     KC_VOLD,   KC_VOLU,   KC_NO,   KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   KC_NO,     RGB_SAD,   RGB_SAI,   KC_NO,     KC_NO    },
-                  {   KC_NO,     KC_NO,     KC_NO,     RGB_TOG, KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   RGB_RMOD,  RGB_MOD,   KC_NO,     KC_NO,     RGB_VAI  },
-                  {   KC_GRV,    KC_NO,     KC_NO,     KC_NO,   KC_NO,    KC_NO,   RESET,    KC_NO,   KC_NO,   KC_NO,     KC_NO,     KC_NO,     RGB_HUD,   RGB_VAD    }
+      [_FN2] = {  {   KC_GESC,   KC_F1,     KC_F2,     KC_F3,   KC_F4,    KC_F5,   KC_F6,    KC_F7,   KC_F8,   KC_F9,     KC_F10,    KC_F11,    KC_F12,    KC_NO,     KC_PGUP    },
+                  {   KC_NO,     RGB_M_P,   RGB_M_B,   RGB_M_R, RGB_M_SW, RGB_M_T, RGB_SPD,  RGB_SPI, KC_NO,   KC_NO,     KC_NO,     KC_NO,     KC_NO,     KC_NO,     KC_PGDOWN  },
+                  {   KC_NO,     KC_VOLD,   KC_VOLU,   KC_NO,   KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   KC_NO,     RGB_SAD,   RGB_SAI,   KC_NO,     KC_NO,     KC_PGUP    },
+                  {   KC_NO,     KC_NO,     KC_NO,     RGB_TOG, KC_NO,    KC_NO,   KC_NO,    KC_NO,   KC_NO,   RGB_RMOD,  RGB_MOD,   KC_NO,     KC_NO,     RGB_VAI,   KC_PGDOWN  },
+                  {   KC_GRV,    KC_NO,     KC_NO,     KC_NO,   KC_NO,    KC_NO,   RESET,    KC_NO,   KC_NO,   KC_NO,     KC_NO,     KC_NO,     RGB_HUD,   RGB_VAD,   RGB_HUI    }
                }
 };
 
diff --git a/keyboards/keychron/k7/rgb/ansi/ansi.c b/keyboards/keychron/k7/rgb/ansi/ansi.c
index b22ae836c6..6bd559605f 100644
--- a/keyboards/keychron/k7/rgb/ansi/ansi.c
+++ b/keyboards/keychron/k7/rgb/ansi/ansi.c
@@ -23,11 +23,11 @@
 __attribute__ ((weak))
 led_config_t g_led_config = {
     {
-        {   0,       1,       2,       3,       4,       5,       6,       7,       8,       9,      10,      11,      12,      13 },
[... 18 lines cut ...]
 /* key matrix size */
 #define MATRIX_ROWS 5
-#define MATRIX_COLS 14
+#define MATRIX_COLS 15
 
 #define DIODE_DIRECTION COL2ROW
 
-#define MATRIX_COL_PINS { C0, C2, A11, B3, B4, B5, B6, B7, B8, B9, B10, B11, B12, B13 }
-#define MATRIX_ROW_PINS { A10, D11, D10, D9, D8 }
+#define MATRIX_COL_PINS { A8, A9, A10, A11, A12, A13, A14, A15, B0, B1, B2, B3, B4, B5, B6 }
+#define MATRIX_ROW_PINS { D11, D10, D9, D8, D7 }
 
 // Connects each switch in the dip switch to the GPIO pin of the MCU
 #define DIP_SWITCH_PINS { D6, D5 }
@@ -48,7 +48,7 @@
 #define LED_MATRIX_ROW_CHANNELS 3
 #define LED_MATRIX_ROWS_HW (LED_MATRIX_ROWS * LED_MATRIX_ROW_CHANNELS)
 //note how ROW pins are assined in the following sequence top down row1 RBG, row2 RBG and so on
-#define LED_MATRIX_ROW_PINS { A7, C5, C8, C11, A9, A6, C4, C7, C10, A8, C15, C3, C6, C9, D10 }
+#define LED_MATRIX_ROW_PINS { C4, C6, C5, C7, C9, C8, C10, C12, C11, C13, B13, C14, D3, B14, B15 }
 
 #define LED_MATRIX_COLS MATRIX_COLS
 #define LED_MATRIX_COL_PINS MATRIX_COL_PINS
`
anywone having same issue ?

dexter93, 2022-12-23:
The initial bringup was on a optical k7 rgb and confirmed working. Maybe there is a pcb revision. Will update when things become a bit clearer

dexter93, 2023-01-06:
K7 RGB hotswap mechanical got hit by silicon shortage, hence the 247b chip inside. Soldered was not

the rest of the K7 RGB variants are using the 248b chip, so things related to this PR should be moved accordingly to a `hotswap` variant

closed 2023-10-21 by dexter93