SonixQMK/qmk_firmware #278: [Bug] Keychron K6 Layout won't load in VIA

billyjacoby, 2022-07-31:
## Describe the Bug

## System Information

 - Keyboard:  Keychron K6
   - Revision (if applicable):  RGB ANSI
 - Operating system: macOS
 - AVR GCC version: 8.5.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.7.101

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I've been able to flash the QMK firmware and everything works for my Keychron K6 but I cannot load the layout in VIA.

VIA will recognize the keyboard and seems like it connects to it, but the layout is never loaded and as such I can't make any changes

tbowmo, 2022-10-27:
Do you sideload the correct json file for your K6 in via? Out of the box via doesn't recognize any of the keyboards in this repo, so you have to create / get a json file with correct setup for your keyboard, and use that in via.

pcs3rd, 2022-11-01:
This sounds similar to what I'm seeing with my keychain c1 white.
I can load the via configuration without issue and the design tab shows the layout, but once I select the configure tab, I just get "Chippy" with "loading" under it.

I have 3 errors under dev tools so far.   

```index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee: Array(3)0: 01: 82: 128length: 3[[Prototype]]: Array(0) Bad Resp: Uint8Array(32)
_hidCommand @ index.309befdef3a90fd3b847.js:2
index.309befdef3a90fd3b847.js:2 Uncaught (in promise) Error: Receiving incorrect response for command
    at kn._hidCommand (index.309befdef3a90fd3b847.js:2:51130)
    at async kn.flushQueue (index.309befdef3a90fd3b847.js:2:50521)
index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee: Array(3) Bad Resp: Uint8Array(32)```

davecarrijo, 2022-12-14:
> This sounds similar to what I'm seeing with my keychain c1 white.
> 
> I can load the via.json configuration without issue and the design tab shows the layout, but once I select the configure tab, I just get "Chippy" with "loading" under it.
> 
> I tried making a new layout under KLE to confirm the json data was good. Same issues.
> 
> 
> 
> I have 3 errors under dev tools so far, of which I don't seem to get for my raspi-pico keyboard with qmk under VIA 
> 
> 
> 
> Keychron c1 w/ sonixQMK  from source (win10)
> 
> ```
> 
> VM47:37274 WebXR emulator extension overrides native WebXR API with polyfill.
> 
> index.309befdef3a90fd3b847.js:2 412350 412350
> 
> /#/design:1 The resource https://usevia.app/images/chippy.png was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
> 
> index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee Array(2) Correct Resp: Uint8Array(32)
> 
> index.309befdef3a90fd3b847.js:784 Setting connected device: 9 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee Object
> 
> index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee Array(2) Correct Resp: Uint8Array(32)
> 
> index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee: Array(3)0: 01: 82: 128length: 3[[Prototype]]: Array(0) Bad Resp: Uint8Array(32)
> 
> _hidCommand @ index.309befdef3a90fd3b847.js:2
> 
> index.309befdef3a90fd3b847.js:2 Uncaught (in promise) Error: Receiving incorrect response for command
> 
>     at kn._hidCommand (index.309befdef3a90fd3b847.js:2:51130)
> 
>     at async kn.flushQueue (index.309befdef3a90fd3b847.js:2:50521)
> 
> index.309befdef3a90fd3b847.js:2 Command for 0f1f3a5c-a616-4a13-a3ae-3b85c40f21ee: Array(3) Bad Resp: Uint8Array(32)
> 
> _hidCommand @ index.309befdef3a90fd3b847.js:2
> 
> ```
> 
>   
> 
> Upstream QMK on raspi-pico from source:   
> 
> ```
> 
> VM597:37274 WebXR emulator extension overrides native WebXR API with polyfill.  
> 
> index.309befdef3a90fd3b847.js:2 412350 412350
> 
> ```
> 
> Actually, Now that I'm home and on my linux system, I can't seem to reproduce that error, but am having the same issue
> 
> 

News?