SonixQMK/qmk_firmware #304: [Bug] Keychron K3 locks up or spam keys

card78, 2022-11-24:
## Describe the Bug

## System Information

 - Keyboard:  
   - Revision (if applicable):  Keychron K3 V2 RGB optical ANSI
 - Operating system: Compiled on Mac. Tested on Mac and Windows.
 - AVR GCC version: 8.5.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.7.101

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
This is related to [https://github.com/SonixQMK/qmk_firmware/issues/271](#271)
If you revert [https://github.com/SonixQMK/qmk_firmware/commit/238e200904f2ccea5de10bddb70ac07845e5830d](238e200), the bug goes away.

patriksh, 2023-01-28:
@card78 any updates on your end? I'm having a similar issue.

starptr, 2023-03-25:
Reverting `238e200` no longer fixes this issue; I guess there's been new commits to `sn32_master` branch of the chibios-contrib repo that causes this

sambartle, 2023-05-31:
I'm currently trying to support a new keyboard and I believe I might be having a similar issue.. 

I get the same thing.. if you press multiple keys, youll fairly quickly get either a lock up, or one of the keys you pressed just repeating constantly until you unplug the keyboard. 

I've tried reverting the change and pulling the sn32_master branch (as well as the sn32 one) and still have the issue though. 

The linked issue refers to a timer issue.. anyone have any details on exactly what that issue is so i might try and debug this please?

humbertogontijo, 2023-07-26:
Is the k3 pro anything related to k3v2? If so, a lot could be taken from https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k3_pro

patriksh, 2023-07-27:
Correct me if I'm wrong, but I believe the K3 Pro has a completely different MCU (STM32 on the Pro, SN32 on the V2).

dexter93, 2023-10-21:
See https://github.com/SonixQMK/qmk_firmware/issues/336

closed 2023-10-21 by dexter93