SonixQMK/qmk_firmware #305: [Feature Request] Add Vial Support for the Keychron K2,  (and possibly to other keyboards as well).

Harlequin69, 2022-11-27:
Add VIAL support to keyboards in this repo. Especially the Keychron K2, since it's what I'm using.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
SonixQMK already has the necessary files to support VIA for the Keychron K2 /K2V2 and other keyboards, VIAL is an opensource alternative to VIA and has an added benifit of storing the config on the board itself, so loading the json file everytime is not necessary. Also It works offline so it is a good option if the via website is down or internet is not working for some people.

zench1k, 2023-01-07:
is it possible to support k2v2 white leds in the future?

nikolmedo, 2023-01-31:
Is it possible to support k5 RGB leds in the future?

closed 2023-07-23 by github-actions[bot]