SonixQMK/qmk_firmware #307: [Bug] LED flickering with new barnches | Redargon K556 Devarajas

jpeter01, 2022-11-27:
After compiling and flashing the firmware from an up to date branch (sn32_master_openrgb), there is a barely but noticeable LED flickering on the K556. With an older version (sn32_openrgb) there isn't any visible flickering.

I checked the flickering also with a phone camera, and on the cam the sn32_master_openrgb version made a clearly visible flickering, but there was no noticeable flicker on the cam with the sn32_openrgb version. The flickering can be also seen without any camera, if you aren't looking directly on the keyboard.

## Describe the Bug

## System Information

 - Keyboard:  Redragon K556RGB
   - Revision (if applicable):  
 - Operating system: openSUSE Leap 15.4
 - AVR GCC version: 12

 - ARM GCC version: 12

 - QMK Firmware version: 0.7.101

 - Any keyboard related software installed? No.
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Wallvon, 2024-07-03:
I am having the same issue with the Redragon K582. In https://github.com/SonixQMK/qmk_firmware/pull/347 I also saw mentions of the LEDs flickering. I'll be doing some testing to see if I can find a solution, but seeing as I am quite new to QMK I don't really know where to look.

Wallvon, 2024-07-03:
Update: I am also working on VIA support for the K582, but I've noticed that the only RGB effects that work are None (so disabled) and solid colour.