SonixQMK/qmk_firmware #314: [Bug] Keyboard spams /s3//s3//s3//s3/...

Raphilator, 2023-01-09:
## Describe the Bug
My keyboard sometimes spams /s3//s3//s3//s3//s3//s3//s3/... and it doesn't stop. I don't now when it happens (it just happens randomly) and i have to replug my keyboard to stop it.

## System Information
 
 - Keyboard:  Redragon K580 Vata
 - Operating system: Windows 11
 - AVR GCC version: 8.5.0
 - ARM GCC version: 10.1.0
 - QMK Firmware version: 0.15.12

 - Any keyboard related software installed? 
   - [x] AutoHotKey
   - [ ] Karabiner
   - [x] Other: SignalRGB

## Additional Context
The Media Key/Macro Key Leds don't work and i can't even use the macro keys.

Logun147, 2023-06-04:
I have the same issue but its b3 on my gmmk 2

dexter93, 2023-10-21:
See https://github.com/SonixQMK/qmk_firmware/issues/336

closed 2023-10-21 by dexter93