SonixQMK/qmk_firmware #324: [Bug] Keychron K8 RGB ISO crash/lock-up

deadproprietary, 2023-02-22:
## Describe the Bug

My Keychron K8 RGB ISO locks up permanently after **typing** for a random amount of keystrokes and becomes unresponsive. It does not seem to be caused by the amount of keys pressed at once. During the lockup, the keyboard may send input of the last pressed key as if it were held, and random RGB backlights may flicker very rapidly.  
Re-plugging the keyboard either through USB or the switch resets it.

## System Information

 - Keyboard:  Keychron K8_UK RGB V2 Gateron
 - Operating system: Windows 10 
 - AVR GCC version: 5.4.0
 - ARM GCC version:  9.2.1
 - QMK Firmware version: 0.7.101
 - Any keyboard related software installed? 
   - [X] AutoHotKey 
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I have tested this both off the `sn32_master` and `sn32_develop` branches. Both exhibit the same issue.

dexter93, 2023-02-23:
old ARM GCC versions prior to 10.3 stable were prone to introducing this. Please test with the latest `gcc-arm-none-eabi`

deadproprietary, 2023-02-23:
issue's been completely resolved and i'm typing this with no problems from the firmware! thank you!

closed 2023-02-23 by deadproprietary

davecarrijo, 2023-02-24:
> old ARM GCC versions prior to 10.3 stable were prone to introducing this. Please test with the latest `gcc-arm-none-eabi`

Some updates into BT mister Dexter?

dexter93, 2023-02-24:
> > old ARM GCC versions prior to 10.3 stable were prone to introducing this. Please test with the latest `gcc-arm-none-eabi`
> 
> Some updates into BT mister Dexter?

replied on the relevant issue