SonixQMK/qmk_firmware #325: [Feature Request] Add support for the Keychron K3 (White)

danieljharris, 2023-03-05:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

There is already support for the Keychron K3 RGB keyboard, but there seems to be no support for the white model yet.

Is there any documentation on how to convert/port from Keychron RGB versions to White versions? More than happy to try and implement it myself, just need to be pointed in the right direction on how to do it.

Many thanks!

closed 2023-07-23 by github-actions[bot]

HubKing, 2023-11-11:
I came here to see if K3 White is supported. Too bad that it is not supported. I don't know how QMK works, so I am just guessing but, shouldn't the RGB version be more complex since it has 3 LED's compared to 1 LED that the white version has? I wonder why white LED version is not supported when RGB version is. Is it simply because almost no one is using it?

danieljharris, 2023-11-11:
@HubKing I think the white and RGB versions are roughly the same difficulty to add QMK support for but they both have different chips inside so support needs to be added separately for each.

And yeah I'm guessing most people who want to add QMK support get the RGB version sadly.

HubKing, 2023-11-12:
We are talking about this "White backlight" version, 

[image]

not this "White, non-backlight" version,

[image]

right?

danieljharris, 2023-11-12:
Correct, I've got the first one (White backlight). I believe all 3 (White backlight, RGB backlight, and white non-backlight) have different chips.