SonixQMK/qmk_firmware #331: too large to run in SRAM

SSSLiurong, 2023-06-06:
It seems that my program is too large to run in SRAM, but I didn't add any other extra function except for basic input function. why?
I use SN32f268 as my mcu. 
[image]

dexter93, 2023-06-20:
See https://github.com/SonixQMK/qmk_firmware/blob/sn32_develop/keyboards/handwired/onekey/sn32f260/rules.mk

fightforlife, 2023-09-19:
I am also using 268f with sonixqmk and a slimmed RGB driver (no double buffer, no idle thread, minimized stack,..)
See here for reference
https://github.com/fightforlife/qmk_firmware/tree/sn32_develop_k66v1/keyboards/womier/k66

closed 2023-10-21 by dexter93