SonixQMK/qmk_firmware #348: [Bug] Keychron K8: Backlight flashes and flickers column on key press

warhuhn, 2023-08-06:
### Describe the Bug

So, if i'm in plain backlight mode, and press down a key, the backlight for that column's keys flickers.

It's much more noticable on the left side of the keyboards. For example, if i press the left control key, i can very clearly see the backlight for the escape key flicker.

The strength of the flickering fades gradually for each columns towards the right, but is still very faintly present, if you look very close.

At first i thought i was stuck in a key press mode, but i've used `#define DISABLE_RGB_MATRIX_...` to disable all modes except plain in the `config.h` file and the issue persists.

### Keyboard Used

qmk compile -kb keychron/k8/rgb/iso -km iso_rgbtest

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ Found qmk_firmware at D:/projects/qmk_firmware.
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: D:/projects/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: sn32_develop
Ψ Repo version: 0.7.101
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-21 10:22:38 +0200 --  (5b4836ca)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

warhuhn, 2023-08-06:
I just checked, the issues present before #344 was merged.

warhuhn, 2023-08-06:
If i switch to software pwm, by adding 

```c
#define SN32_PWM_CONTROL SOFTWARE_PWM
```

to my keymap's `config.h`, the problem becomes very visible. The entire key row lights up white on keypress.

I also tried disabling the keypress feature by adding

```c
#ifdef RGB_MATRIX_KEYPRESSES
#undef RGB_MATRIX_KEYPRESSES
#endif

#ifdef RGB_MATRIX_KEYRELEASES 
#undef RGB_MATRIX_KEYRELEASES
#endif
```

That disabled the keypress rgb effect modes, but doesn't prevent the rows lighting up.

warhuhn, 2023-08-06:
Just reverted back to the original, official firmware as a sanity check that i didnt break anything, and LEDs worked as they should.

dexter93, 2023-08-07:
sounds like a bug. However,

```c
Ψ Found arm-none-eabi-gcc version 10.1.0
```
that version is known to be buggy, please bump to anything over 10.3+

If that doesn't fix the issue, can you try
```c
#define MATRIX_UNSELECT_DRIVE_HIGH
```
and see if the issue persists?

warhuhn, 2023-08-08:
```c
#define MATRIX_UNSELECT_DRIVE_HIGH
```

Did indeed fix it!

Does the default configuration need updating? 

Or is my board on a different revision?
It says `Keychron K8_UK RGB_FTCB_SFGKB Ver2.0_20200701` on the board.

warhuhn, 2023-08-08:
I'll also try without that option after updating my gcc, as soon as i figure out how to do that in QMK MSYS.

warhuhn, 2023-08-25:
Okay, i've tried compiling without `#define MATRIX_UNSELECT_DRIVE_HIGH` and a newer gcc version.

Problem still persisted, so the configuration probably needs updating. I'll open a pull request in a moment.

```sh
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/suigintou/Projekte/sonix-qmk
Ψ Detected Linux (openSUSE Tumbleweed).
Ψ Git branch: sn32_develop
Ψ Repo version: 0.7.101
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12
Ψ Found avr-gcc version 12
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-21 10:22:38 +0200 --  (5b4836ca)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

closed 2023-09-08 by dexter93