SonixQMK/qmk_firmware #359: [Keyboard] Adding Keychron K2v2 ANSI variant to current active branch.

ITNerd69, 2023-10-31:
### Issue Description

The keychron K2v2 has been supported for a long time on SonixQMK. 
Since the active branch sn32_develop opened, the K2v2's code had not been added/ ported over. 
Recently I saw that the K2v2's ISO variant had been ported over by @KeijoMika to the active branch via the PR #358 .

The same should be done for the ANSI variant as well, since the ANSI variant of the keyboard is much more common than the ISO variant. 

I am willing to do the necessary work and make the pull request in order to port it over. 
Though I can see from @KeijoMika work that quite a few files have been changed/ improved since the first iteration from @dexter93.
 I have some background in C so I'd like to understand the code structure and purpose of the files and make my own improvements on the ANSI variant before porting it over, thus simply copy pasting the code into the new repo isn't enough. 

This will be my first pull request so any help is welcomed :)

KeijoMika, 2023-10-31:
Hi,

It should be fairly easy to modify it to work with ANSI version. I am pretty sure the pins in ANSI version are the same as ISO.

You can find it the ANSI version on an older branch sn32_master:
https://github.com/SonixQMK/qmk_firmware/tree/sn32_master/keyboards/keychron/k2

Then you just need to do similar changes to the info.json:
Use the sn32_master as a reference to do the changes to layout (create layout_ansi instead of layout_iso), rgb matrix and product id. No background in C really needed, just need to follow the logic, of course some programming background will help here. "ansi.c", "ansi.h" from the older branch will be helpful here.

Also matrix led count needs to be 84 for ANSI I believe. Keymap should be pretty much the same as the one in sn32_master.

And for VIA support, you will need to create k2_ansi.json with just some small changes to the layout, which you can test by manually loading in http://usevia.app

ITNerd69, 2023-10-31:
Thanks for reaching out !
I have so far been able to get the firmware files setup in the same folder structure as you. I've setup the info.json, keymap rules.mk and config.h files. I've compiled and flashed the firmware to the keyboard without any errors. Via work too after some modification to the via .json file.
I am facing a few problems though. Hope you'll be able to help.
The Enter key and z key are not working, both are blinking red. The 5th row has one key shifted: shift works as expected, BUT after that the z key does not work, then from there the entire row keys has been shifted, like pressing the X button prints z, pressing the C button prints x and so on and so forth, the ? key does not exist on the keyboard for some reason :/
Although when using via, once I manually set the two keys, then they work.
I'm attaching my info.json file and a few screen shots for clarification. 

Please look into it in your spare time.
Pastebin for [info.json](https://pastebin.mozilla.org/Z1wuxM0R)
[Screenshots](https://imgur.com/a/Rz7gJYN)

KeijoMika, 2023-10-31:
Hi,

Upon quick inspection, in info.json can you try in layout_ansi:

Instead of:
```
{"matrix":[3, 12], "x":12.75, "y":3, "w":2.25},
```

Try:
```
{"matrix":[3, 13], "x":12.75, "y":3, "w":2.25},
```

Also later in the file in layout_ansi also:

Instead of:
```
				{"matrix":[4, 0], "x":0, "y":4, "w":2.25},
				{"matrix":[4, 3], "x":2.25, "y":4},
				{"matrix":[4, 4], "x":3.25, "y":4},
				{"matrix":[4, 5], "x":4.25, "y":4},
				{"matrix":[4, 6], "x":5.25, "y":4},
				{"matrix":[4, 7], "x":6.25, "y":4},
				{"matrix":[4, 8], "x":7.25, "y":4},
				{"matrix":[4, 9], "x":8.25, "y":4},
				{"matrix":[4, 10], "x":9.25, "y":4},
				{"matrix":[4, 11], "x":10.25, "y":4},
				{"matrix":[4, 12], "x":11.25, "y":4},
				{"matrix":[4, 13], "x":12.25, "y":4, "w":1.75},
				{"matrix":[4, 14], "x":14, "y":4},
				{"matrix":[4, 15], "x":15, "y":4},

```
Try:

				{"matrix":[4, 0], "x":0, "y":4, "w":2.25},
				{"matrix":[4, 2], "x":2.25, "y":4},
				{"matrix":[4, 3], "x":3.25, "y":4},
				{"matrix":[4, 4], "x":4.25, "y":4},
				{"matrix":[4, 5], "x":5.25, "y":4},
				{"matrix":[4, 6], "x":6.25, "y":4},
				{"matrix":[4, 7], "x":7.25, "y":4},
				{"matrix":[4, 8], "x":8.25, "y":4},
				{"matrix":[4, 9], "x":9.25, "y":4},
				{"matrix":[4, 10], "x":10.25, "y":4},
				{"matrix":[4, 11], "x":11.25, "y":4},
				{"matrix":[4, 13], "x":12.25, "y":4, "w":1.75},
				{"matrix":[4, 14], "x":14, "y":4},
				{"matrix":[4, 15], "x":15, "y":4},

Next in rgb matrix layout:
Instead of:
```
{ "matrix": [ 3, 12 ], "flags": 4, "x": 166, "y": 38 },
```

Try:
```
{ "matrix": [ 3, 13 ], "flags": 4, "x": 166, "y": 38 },
```

Later in rgb matrix layout instead of:
```
				{ "matrix": [ 4, 0 ], "flags": 4, "x": 8, "y": 51 },
				{ "matrix": [ 4, 3 ], "flags": 4, "x": 28, "y": 51 },
				{ "matrix": [ 4, 4 ], "flags": 4, "x": 40, "y": 51 },
				{ "matrix": [ 4, 5 ], "flags": 4, "x": 53, "y": 51 },
				{ "matrix": [ 4, 6 ], "flags": 4, "x": 65, "y": 51 },
				{ "matrix": [ 4, 7 ], "flags": 4, "x": 78, "y": 51 },
				{ "matrix": [ 4, 8 ], "flags": 4, "x": 90, "y": 51 },
				{ "matrix": [ 4, 9 ], "flags": 4, "x": 103, "y": 51 },
				{ "matrix": [ 4, 10 ], "flags": 4, "x": 115, "y": 51 },
				{ "matrix": [ 4, 11 ], "flags": 4, "x": 128, "y": 51 },
				{ "matrix": [ 4, 12 ], "flags": 4, "x": 140, "y": 51 },
				{ "matrix": [ 4, 13 ], "flags": 4, "x": 157, "y": 51 },
				{ "matrix": [ 4, 14 ], "flags": 4, "x": 174, "y": 51 },
				{ "matrix": [ 4, 15 ], "flags": 4, "x": 187, "y": 51 },
```
Try:
```
				{ "matrix": [ 4, 0 ], "flags": 4, "x": 8, "y": 51 },
				{ "matrix": [ 4, 2 ], "flags": 4, "x": 28, "y": 51 },
				{ "matrix": [ 4, 3 ], "flags": 4, "x": 40, "y": 51 },
				{ "matrix": [ 4, 4 ], "flags": 4, "x": 53, "y": 51 },
				{ "matrix": [ 4, 5 ], "flags": 4, "x": 65, "y": 51 },
				{ "matrix": [ 4, 6 ], "flags": 4, "x": 78, "y": 51 },
				{ "matrix": [ 4, 7 ], "flags": 4, "x": 90, "y": 51 },
				{ "matrix": [ 4, 8 ], "flags": 4, "x": 103, "y": 51 },
				{ "matrix": [ 4, 9 ], "flags": 4, "x": 115, "y": 51 },
				{ "matrix": [ 4, 10 ], "flags": 4, "x": 128, "y": 51 },
				{ "matrix": [ 4, 11 ], "flags": 4, "x": 140, "y": 51 },
				{ "matrix": [ 4, 13 ], "flags": 4, "x": 157, "y": 51 },
				{ "matrix": [ 4, 14 ], "flags": 4, "x": 174, "y": 51 },
				{ "matrix": [ 4, 15 ], "flags": 4, "x": 187, "y": 51 },
```

Hopefully this will fix at least some of the issues. If some remain, I can try to help some more.

ITNerd69, 2023-10-31:
Thanks !! I just applied the fixes you suggested and all the issues have been resolved !
Furthermore, I added, the rgb sleep feature in the config.h file, so now the keyboard RGB goes to sleep along with the OS ! I'll clean everything up and submit a pull request if nothing else breaks :)

ITNerd69, 2023-10-31:
@KeijoMika I submitted the pull request. If you happen to find any issues then please let me know :)

ITNerd69, 2023-11-03:
Umm, it's been two days. @dexter93 already approved the changes. Why isn't the pull request not merged yet ? Am I missing something ? I'm still quite new to this so there might be something I might've missed.

ITNerd69, 2023-11-04:
Pull request has been merged ! Closing this issue as solved.

closed 2023-11-04 by ITNerd69