SonixQMK/qmk_firmware #377: [Help] Reverse Cherry MX 10.0N RGB QMK

Metaln00b, 2024-03-27:
Hello,

i want to reverse the Cherry MX 10.0N RGB to use QMK.
Can anyone help me with this?

I have the following information about the keyboard.

MCU: E VISION VS11K09A-1 Seems to be based on the Sonix SN32F248BF [Reference](https://github.com/MartinNievas/K530)
LED: 3 * VS12L12A
Board: Rev. 3

Is there firmware for testing the pins to create a mapping?
I would be grateful for any tips.
[image]

as32, 2024-04-25:
I reversed a very similar keyboard.
Checked the mapping in the stock firmware of the 10.0N
Cols: C0, C1, -, C3 ... C14, A8 ... A14
Rows: C15, D11, D10, D9, D8, D7

Metaln00b, 2024-04-26:
Thank you 🙏

How did you get the stock firmware? I was not able to get the firmware from the Keyboard neither from the cherry tool.

as32, 2024-04-26:
> Thank you 🙏
> 
> How did you get the stock firmware? I was not able to get the firmware from the Keyboard neither from the cherry tool.

Did you ever look for it in C:\Program Files (x86)\Cherry\UtilitySoftware\UpdateData ? 😉

Metaln00b, 2024-04-26:
> > Thank you 🙏
> > 
> > How did you get the stock firmware? I was not able to get the firmware from the Keyboard neither from the cherry tool.
> 
> Did you ever look for it in C:\Program Files (x86)\Cherry\UtilitySoftware\UpdateData ? 😉 

Oh maybe i use a newer version. It is Qt5 based an the path is C:\Program Files (x86)\Cherry Utility Software\cus.exe

There is no UpdateData folder 😅

as32, 2024-04-26:
> Oh maybe i use a newer version. It is Qt5 based an the path is C:\Program Files (x86)\Cherry Utility Software\cus.exe

Ok, I wasn't aware of that. 😉 
In an older version, they were just in there in several exe files and easy to extract.
 
> There is no UpdateData folder 😅

That's bad luck. 😅

as32, 2024-04-27:
The code is going to be very similar to the keyboard which I implemented recently.
So I thought I might start a branch based on that.
The branch would be here:
https://github.com/as32/qmkCherryDev/tree/Issue377__Add__Cherry_mx_10.0n_rgb

as32, 2024-04-29:
Meanwhile, I pushed the changes and I compiled and flashed it a few times without problems.
Seems to be usable (at your own risk 😉 ):

https://github.com/SonixQMK/qmk_firmware/assets/5454760/c023a02a-50fb-41b1-b35a-a242de4ee926

as32, 2024-04-29:
Further development changes might follow... (be careful 😉 )

Metaln00b, 2024-04-29:
> Meanwhile, I pushed the changes and I compiled and flashed it a few times without problems.
> Seems to be usable (at your own risk 😉 ):
> 
> https://github.com/SonixQMK/qmk_firmware/assets/5454760/c023a02a-50fb-41b1-b35a-a242de4ee926
> 
> 

It is working great. I need to add German Layout but. Wow 🙏🙏🙏

Thank you 🤘

One thing we need is to add 046A:00DF to the sonix flasher 😋

closed 2024-08-28 by github-actions[bot]