SonixQMK/qmk_firmware #402: [Bug] QMK failiure on Python 3.12. for `sn32_stable` due to https://github.com/qmk/qmk_firmware/pull/22562

Swiftb0y, 2024-09-19:
### Describe the Bug

Since https://github.com/qmk/qmk_firmware/pull/22562 has not been included, using this fork on newer python versions fails. Please merge an recently up-to-date upstream commit into `sn32_stable` if possible to fix this.

### Keyboard Used

N.A.

### Link to product page (if applicable)

N.A.

### Operating System

Fedora 40

### qmk doctor Output

```text
N.A.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

None

dexter93, 2024-09-19:
Closing since `sn32_stable` is an inactive branch left here for archival purposes. Please move to `sn32_develop`

closed 2024-09-19 by dexter93

Swiftb0y, 2024-09-19:
Thanks for the clarification. Will do.

dexter93, 2024-09-20:
Bug fixed with ca5b5ee0f6098f7920ed31a40b5fb1b4bd673515