SonixQMK/qmk_firmware #420: mappings present in `sn32_master` missing from `sn32_develop`

Swiftb0y, 2024-09-20:
There are a couple mappings present in `sn32_master` (such as the keychron K3) which are missing from the `sn32_develop` branch. Is there a reason for that? Whats the relation between these branches exactly? Is the divergence between these two branches intentional?

dexter93, 2024-09-21:
`sn32_master` / `sn32_stable`  / `sn32`: older trees, functional but hacked up approach. Significantly harder to maintain. status: abandoned/archived.
`sn32_develop`: rolling release based on `qmk/develop`, updated quarterly ( or when beneficial patches get merged upstream). Scalable, portable, minimally intrusive to upstream, easier to maintain. Currently the active tracking branch.

Missing mappings/keyboards: Maintainer issue. We dropped/didn't merge them to `sn32_develop` because significant changes mean active maintainers are needed to verify functionality and keep up with updating. We actively encourage people to migrate mappings from `sn32*` to `sn32_develop` and the process is easy, but it only makes sense if having the hardware to verify they work reliably.

Swiftb0y, 2024-09-21:
> We actively encourage people to migrate mappings from sn32* to sn32_develop and the process is easy, but it only makes sense if having the hardware to verify they work reliably.

Thanks. I'll look into doing that for the keychron K3. Is there more documentation (better than the 5000 commit log diff) on what exactly has changed between these branches and whats needed to migrate from the older trees to develop?

dexter93, 2024-09-22:
> Thanks. I'll look into doing that for the keychron K3. Is there more documentation (better than the 5000 commit log diff) on what exactly has changed between these branches and whats needed to [...]

The bulk of the changes are covered by the [data driven migration](https://github.com/SonixQMK/qmk_firmware/blob/sn32_develop/docs/reference_info_json.md) and for RGB, the [sn32f2xx](https://github.com/SonixQMK/qmk_firmware/blob/sn32_develop/docs/drivers/sn32f2xx.md) RGB driver. You can see current configuration examples for sn32 [here](https://github.com/search?q=repo%3ASonixQMK%2Fqmk_firmware+%22bootloader%22%3A+%22sn32-dfu%22+path%3A%2F%5Ekeyboards%5C%2F%2F&type=code)