SonixQMK/qmk_firmware #432: [Bug] RGB backlight for Redragon K556 doesn't show correct colors

zinapra, 2024-11-19:
### Describe the Bug

Using firmware from s32_develop branch. When keyboard is set to white or yellow LED's glow red. Lowering the brightness doesn't fix the issue.

### Keyboard Used

redragon/k556

### Operating System

Win 11

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

NLTD2010, 2025-01-30:
I have an Aukey KM-G14 and have the same issue.

NLTD2010, 2025-02-07:
Update: When using the old rgb matrix code in the sn32_master branch, everything works fine!