SonixQMK/qmk_firmware #442: [Feature Request] What about HFD80CP100?

frsvme, 2025-02-13:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi! I buyed DarkProject KD87A+Numpad [DP-KD-87A-000101-GMS] (As far as I understand, this is a separate version for the CIS market, with a magnetic mount of the numpad unit.). Native software is a tin can. I would really like to switch to qml, so I disassembled the keyboard to identify the controller. It turned out to be the HFD80CP100. Unfortunately, I'm stuck on this and don't know what to do next... The numpad unit is defined as a separate HID device, but there are no questions about it yet. I want to flash the main keyboard specifically.

[image]

[image]

dexter93, 2025-02-18:
`HFD80CP100` is actually `SN32F299F`. See https://github.com/SonixQMK/qmk_firmware/pull/436

frsvme, 2025-02-19:
> `HFD80CP100`на самом деле `SN32F299F`. Смотрите [#436](https://github.com/SonixQMK/qmk_firmware/pull/436)

Does this mean that I can compile the mc and use it on my keyboard? I'm sorry, but I just don't understand a bit...
and what are the limitations of my platform at the moment, if its firmware is possible? Will the backlight of the keys, the screen, the volume control (encoder) work?

dexter93, 2025-02-20:
> > `HFD80CP100`на самом деле `SN32F299F`. Смотрите [#436](https://github.com/SonixQMK/qmk_firmware/pull/436)
> 
> Does this mean that I can compile the qmk and use it on my keyboard? I'm sorry, but I just don't understand a bit... and what are the limitations of my platform at the moment, if its firmware is possible? Will the backlight of the keys, the screen, the volume control (encoder) work?

Platform support for the chip is linked - you're on your own to figure out the board internals, wiring and drivers needed for functionality.

dexter93, 2025-06-09:
Closing as completed