SonixQMK/qmk_firmware #444: Compiled firmware SN32B248 (without "b" in end)

NastyaLetova, 2025-04-15:
### Issue Description

I have keyboard Jagor T87-P3, and i compared the pinout Redragon k552 it matches.
But differnt version Redragon K552 have SN32B248 and another chip SN32B248b. But in develop branch don't have version for SN32B248 (without "b" in end). How need to change config files to compile firmware for chip without "B"?

I break my keyboard, and i dont have dump (i dont have ST-LINK).

RSATom, 2025-04-29:
Are you sure key matrix uses the same pins? SN32B248 has different amount of pins comparing to SN32B248B. Also, it should be easy to run bootloader if you are able to disassemble it.

RSATom, 2025-04-29:
nvm, I see you already have discussion on discord...

NastyaLetova, 2025-04-29:
> [image]
> [image]
> Are you sure key matrix uses the same pins? SN32B248 has different pinout comparing to SN32B248B. Also, it should be easy to run bootloader if you are able to disassemble it.

VS11K09A. Matrix look like redragon k552 rev1 but use COL2ROW. In old branch which can compile firmware for non B chip all keyboards with non B chip has separate matrix.c file. 
And i don’t understand how to do matrix file for COL2ROW

RSATom, 2025-04-29:
First of all you need to find out all pins related to cols and to rows. The best way to do it is using something like https://en.wikipedia.org/wiki/Multimeter. Do you know what is it and how to use it?

RSATom, 2025-04-29:
Also, if it's COL2ROW - it should use diode between switch and row pin, so you'll have to find diodes related to all rows.

NastyaLetova, 2025-05-02:
> Also, if it's COL2ROW - it should use diode between switch and row pin, so you'll have to find diodes related to all rows.

[image]

[image]

I think it's col2row because each pin in the column is connected to diodes, and have a common pin on each diode in the column

NastyaLetova, 2025-05-02:
[image]
diode pins connect

RSATom, 2025-05-02:
I have a feeling it's row2col actually.

NastyaLetova, 2025-05-02:
> I have a feeling it's row2col actually. Can you please confirm diodes connected to each other in vertical direction (on side with black line)?

yes, i check this using multimeter

NastyaLetova, 2025-05-02:
> > I have a feeling it's row2col actually. Can you please confirm diodes connected to each other in vertical direction (on side with black line)?
> 
> yes, i check this using multimeter

black line on diodes connect to one pin on mcu

RSATom, 2025-05-02:
then it's definitely row2col.  and if you have multimeter just find out all pins connected to cols, and all pins connected to col - then you'll be able to get all keys working. And what exactly mcu your keyboard is using?

closed 2025-10-05 by github-actions[bot]